NCSU-High-Powered-Rocketry-Club/AirbrakesV2 #3: Add testing suite

harshil21, 2024-09-08:
In order to confidently make changes to the code without worrying if something will be broken, I propose to add unit tests to ensure everything remains functional.

[Pytest](https://docs.pytest.org/en/stable/) is the industry standard for writing unit tests. We can then move the `test_*.py` files we currently have into a `tests` directory and give them a clear purpose. The purpose of  "Mock" files in https://github.com/NCSU-High-Powered-Rocketry-Club/AirbrakesV2/tree/simulations can also be addressed with the help of this.

Thoughts @JacksonElia?

Aman-Anas, 2024-09-08:
I would say to leave unit tests for later when the system is more mature. Right now we're still figuring out what the full architecture will look like (classes, interfaces, hardware etc) so we'd have to keep changing tests drastically.

So IMO it's better to put coding time into tuning the system and data analysis/control theory work. If you have a lot of free time there's no problem with making test classes tho

harshil21, 2024-09-08:
> I would say to leave unit tests for later when the system is more mature. Right now we're still figuring out what the full architecture will look like (classes, interfaces, hardware etc) so we'd have to keep changing tests drastically.

Agreed. While I'm prototyping/testing, I don't write unit tests as well, but then there comes a point where unit tests simply help to save time. 

> If you have a lot of free time there's no problem with making test classes tho

I could write some small basic tests next week, depending on the overall progress.

JacksonElia, 2024-09-10:
I agree with @Aman-Anas. I say we hold off on them right now, but it is a good idea. Eventually we should definitely add tests.

closed 2024-09-10 by JacksonElia