NCSU-High-Powered-Rocketry-Club/AirbrakesV2 #7: Use `msgspec` library for higher speed

harshil21, 2024-09-15:
Let's try to use `msgspec` for performance reasons for the `IMUDataPacket` classes. I'll try to benchmark this and see how much of an improvement we get.

Also will need to test if that builds on the Raspberry Pi since I didn't see wheels for it.

_Originally posted by @Aman-Anas in https://github.com/NCSU-High-Powered-Rocketry-Club/AirbrakesV2/pull/4#discussion_r1759957268_

harshil21, 2024-09-15:
Initializing 100,000 `RawDataPackets` before msgspec, 
```
0.03520050000224728
0.030862718995194882
0.031663336005294695
0.03366126600303687
0.032104951998917386
```
After msgspec:
```
0.010517381990212016
0.010180865006987005
0.009927719991537742
0.00970079199760221
0.011575923999771476
```
After taking the average, `msgspec` is 68.25% in initializing these classes!

Pros:
- Fast 
- no boilerplate
- no need of slots since it uses it internally
- no need to define a `__str__`/`__repr__`
Cons:
- no VSCode autocomplete/type hint :(

Overall: let's do it

Aman-Anas, 2024-09-15:
type hints usually work for me, that's odd.

PR #8 (2024-09-15): Use the `msgspec` library for `IMUDataPacket` classes
Closes #7 

To be merged after #6 and testing running/installation/building on the raspberry pi
closed 2024-09-15 by JacksonElia