NCSU-High-Powered-Rocketry-Club/AirbrakesV2 #36: Small issues to be fixed

harshil21, 2024-10-03:
Perfect tasks for newcomers:

- [ ] Round decimal places for the fields `speed`, `avg_acceleration`, and `altitude` in `ProcessedDataPacket`.
- [ ] Create an alias on the raspberry pi to run `source .venv/bin/activate` without typing out the full thing
- [ ] Remove startup script on the raspberry pi.
- [ ] Create new Enum classes in `constants.py` like the exisiting `ServoExtension` class.

Comment on the issue below to get assigned!

Danuragtiwari, 2024-10-04:
Hi @harshil21, 

I’m excited to contribute to this project! Please assign this issue to me, and I’ll start working on it.

Looking forward to collaborating!

Best regards,
Anurag Tiwari

JacksonElia, 2024-10-04:
Out of curiosity, what has you interested in our code?

harshil21, 2024-10-04:
@Danuragtiwari Hi! I took a quick look at your profile and from what I can gather I think you would be well suited to do the last bullet point (mermaid diagram)

Some additional notes on that:

- The mermaid flowchart doesn't have to be exactly like I've drawn it, but it should mention that the Logger and IMU class will be handled in child processes. 

- It would be nice to use bubble shapes for the methods, ovals for classes, and squares for the output of the class (e.g. speed, avg_accel, and altitude)

- Bonus points for making it more beautiful through CSS styles.

Is it possible for you to complete this task within 2 weeks?

Danuragtiwari, 2024-10-04:
@harshil21 Thanks for considering me for this. I'll do my best to complete the task as soon as possible .

harshil21, 2024-10-04:
@Danuragtiwari alright, assigned you. Btw I'm also interested to know why you chose this repository to contribute.

Danuragtiwari, 2024-10-04:
@harshil21, @JacksonElia  I wanted to restart my open-source contributions, and I found this repository. I love problem-solving, so it excited me to join! Thanks for this opportunity :)

closed 2024-11-14 by harshil21