NCSU-High-Powered-Rocketry-Club/AirbrakesV2 #38: More easy tasks

JacksonElia, 2024-10-04:
- [ ] Make landed state trigger properly (you can see in the graph on the readme, it triggers prematurely while the rocket is still falling)
- [ ] Fix/Improve/Add unit tests (the more good quality ones, the better)
- [ ] Find any places in the code that need to be commented better (if you don't understand it after thinking through it moderately hard, it's not good enough)
- [ ] Find any variable, method, or class names that need to be improved (naming is key, maybe some names don't make sense right away)

harshil21, 2024-10-04:
> * [ ]  Make landed state trigger properly (you can see in the graph on the readme, it triggers prematurely while the rocket is still falling)

You mean to just change the constant at which it landedstate is triggered? Right now it considers it landed if it's within 15m of the launch site's height. Set it too low, and it might not trigger if it lands at an higher elevation, set it too high, and it triggers too early. IMO we can go down 7m, but then again I don't know the launch site topography at Huntsville. 

Ideally we would also check the speed and/or acceleration (zero)

JacksonElia, 2024-10-06:
Yeah I wouldn't just change the constant, probably also do something based on accel -- that being said, it's up to whoever does this to test it against multiple data sets and what not to ensure it works

wlsanderson, 2025-04-06:
I feel 3 and 4 were completed with my documentation changes. The unit test and unit test comments will pretty much always be true, so I think this issue can be closed now.

closed 2025-04-08 by wlsanderson