NCSU-High-Powered-Rocketry-Club/AirbrakesV2 #91: Findings and learnings from the Genesis subscale launches

harshil21, 2024-11-17:
We had 2 launches with airbrakes for subscale.

The openrocket simulation said that our target apogee was going to be 524m. 

**Launch 1:** We set `TARGET_ALTITUDE` to `10m`, with the [curve fit convergence threshold](https://github.com/NCSU-High-Powered-Rocketry-Club/AirbrakesV2/blob/a1af870e09fb47e837cf4aaf3dbf7cf022e01712/constants.py#L121) as `[0.1, 0.75]`. This would theoretically ensure that airbrakes deploy right after our apogee converges, which was expected to take <1s (this was done by running the sim (#76) on the pi). 

Findings from Launch 1:  Launch and code worked almost flawlessly. We logged everything we needed, and handled invalid quaternions  correctly. The apogee prediction converged after 0.98 seconds, the same instant where the command to deploy airbrakes was given. It is at the time of writing unclear if our airbrakes physically deployed or not. We hit an apogee of 455.41m. Our predicted apogee right after convergence was 441.664m. Our csv file ended being over 118MB, because we did not correctly detect `LandedState`.

Learnings from Launch 1: After gathering the data, and running the mock sim with the same data on the raspberry pi, with the same constants, we found the convergence time to be 0.02s, which is a massive discrepancy with the actual log file data, which populates the predicted apogee after 0.918s into `CoastState`. 

----

**Launch 2:** Since it was unclear if airbrakes deployed or not, we changed only 3 things for this flight. 
1. Blue tape was put over the airbrakes, which we tested that the fins can cut through easily, while at the same time making sure the wind cannot remove them.
2. Our state changes logic was changed such that the airbrakes were set to never retract when changing between states. This would help us physically see if airbrakes deployed or not when we recovered the rocket.
3. Bonus: We fixed the bug where didn't detect `LandedState`

Findings from Launch 2: While recovering the rocket, it was clear that airbrakes had deployed. Our apogee converged after exactly 1.240 seconds, and our actual apogee was 462.95m. Our predicted apogee reported right after convergence was 451.55m. The flight profile is identical to Launch 1.

Learnings from Launch 2: By looking at the data (accel values), it is still unclear at what point airbrakes deployed (because we would expect a spike in acceleration). `LandedState` was detected correctly this time, which led to a smaller csv file size (20.4 MB). Our apogee prediction convergence time was still unusually high at 1.240 seconds, despite the high convergence threshold which gave us a very low convergence time when we ran the mock. 

----

**Next steps:**

To be completed roughly in order:

- [ ] Upload the cleaned csv files. For launch 1 we need to remove all the extra time sitting in FreeFallState. For both the launches we need to convert the float timestamp back to an integer. (Taken by @harshil21).
- [ ] Investigate when airbrakes might have (not) deployed by employing a Fourier Transform. (Taken by @harshil21)
- [ ] Investigate why our convergence time is not matching what we expect.
- [ ] Do #84. @wlsanderson can you post the pressure, temperature, and launch rod angle readings for both flights?
- [ ] Write (more) unit tests for the recent PR's: #88, #89, #90. We also now need tests for handling of command line arguments and the handling with the display.
- [ ] Push the flight code changes from the Pi, if needed.

wlsanderson, 2024-11-20:
I think for flight 1, its worthwhile to keep a copy stored somewhere (maybe on the google drive) that still has all of the extended free-fall state. 

first flight metadata:
- Pressure: 1016.1 mbar (measurement taken from phone sensor, not IMU)
- Temp: 16c
- Rod angle: 6 degrees

second flight metadata:
- Pressure: 1015.5 mbar (from phone sensor)
- Temp: 15.5c
- Rod angle: 5 degrees

harshil21, 2024-11-21:
We found the problem regarding high convergence time. It's because the IMU processed over a 1000 packets at once, which leads to the curve fit take a longer time than usual. So we need more performance improvements, or artificially limit the number of packets processed (which is a bad idea imo, since we could be very behind on processing data packets from the IMU)

harshil21, 2024-11-27:
Last to-do of the remaining tests will be done in #106. Closing.

closed 2024-11-27 by harshil21