NCSU-High-Powered-Rocketry-Club/AirbrakesV2 #95: Consider upgrading the IMU's mscl library - https://github.com/LORD-MicroStrain/MSCL/releases/tag/v66.0.0. Needs major testing with IMU since the timestamp field is now no longer the unix epoch.

harshil21, 2024-11-20:


JacksonElia, 2024-11-20:
I don't know if this is a great idea because 1. it might be a lot of work, and do we get anything out of it?
2. Our imu is already old as hell so there could be some potential compatibility issues idk

wlsanderson, 2024-11-20:
From the changes you linked:
"Moved Quaternion class to EulerAngles.h, added Rotation class to represent either a Quaternion or EulerAngles object"

Does that just mean that it wont be able to output quaternions anymore and it will just give euler angles?

harshil21, 2024-11-20:
> From the changes you linked: "Moved Quaternion class to EulerAngles.h, added Rotation class to represent either a Quaternion or EulerAngles object"
> 
> Does that just mean that it wont be able to output quaternions anymore and it will just give euler angles?

Not really, it just means that they have a single class which can be a `Quaternion` or `EulerAngles`. Similar to how we have a `IMUDataPacket` which can be `Raw` or `Est`. https://lord-microstrain.github.io/MSCL/Documentation/MSCL%20Documentation/index.html#File:MicroStrain/Inertial/EulerAngles.h:Quaternion

> I don't know if this is a great idea because 1. it might be a lot of work, and do we get anything out of it? 2. Our imu is already old as hell so there could be some potential compatibility issues idk

we can very easily just downgrade it if something doesn't work out

JacksonElia, 2024-11-27:
Wait though what is the benefit of doing this? Seems kinda painful and does it actually get us anything nice? E.g. is it faster somehow?

harshil21, 2024-11-28:
There is no reason to not upgrade, e.g. it has fixes and stuff.

harshil21, 2025-01-16:
upgraded in #126. Tested with the pi and checked the timestamp. Everything's good. Closing.

closed 2025-01-16 by harshil21