NCSU-High-Powered-Rocketry-Club/AirbrakesV2 #108: Code needs to be cleaned up more

JacksonElia, 2024-11-27:
Honestly our code is pretty good at this point, but it could always be better. We need to:
- [ ] Clean up `constants.py`
- [ ] Add comments to the tests (oh boy)
- [ ] Do another go at changing var names and function names to be clearer and convey more info
- [ ] Flesh out `scripts/` a little more, maybe add a nicer UI for them or something
- [ ] Expand the readme, maybe add a little more documentation/guidance where to start for new members

harshil21, 2024-11-28:
Duplicate of #38

wlsanderson, 2024-12-04:
> Duplicate of #38

so real

harshil21, 2024-12-04:
> Flesh out scripts/ a little more, maybe add a nicer UI for them or something

This can probably be done along with/after #106.

closed 2025-02-09 by JacksonElia