NCSU-High-Powered-Rocketry-Club/AirbrakesV2 #152: Remove windows support

JacksonElia, 2025-02-09:
I know I know, I was the one fighting against this happening, but I thought about it more and I'm okay with it. I think that it makes our code a bit uglier, and isn't really that good to have. If we want to onboard a new programmer, part of that is just going to have to be making them get WSL.

Thoughts @harshil21 @wlsanderson @Graicc @MrPezser ?

harshil21, 2025-02-09:
I don't mind removing it. I should also mention that I am preparing initial Cython support, which will require you to have a compiler installed along with ninja/cmake. I don't think Windows has one by default, so that would've definitely necessitated it anyway.

wlsanderson, 2025-02-10:
Yeah I dont really see the point in windows support when the one single "target user" (the Pi) is Linux-based

PR #155 (2025-02-18): Remove Windows support
Closes #152 

Merge after #146
closed 2025-02-18 by wlsanderson