NCSU-High-Powered-Rocketry-Club/AirbrakesV2 #183: Timestamp issues

wlsanderson, 2025-03-29:
In pelicanator launch 1 and 2, there were some weird issues with the delta time not being constant during packets. For some parts of flight, it was higher than usual, meaning we might have not been at a steady 500hz raw/est. This also lines up with the python unix time we logged also. I'll post a graph for both flights, plotting time vs row number, plotting both packet timestamp and unix timestamp (they are pretty much on top of each other so you can't really tell). Black vertical lines identify state changes.

The raspberry pi unix time we recorded also had a different issue. On Pelicanator flight 1, line 19177 in the csv, the raspberry pi unix time left off a number in the last digit, resulting in a time 10x lower than reality. I'm hesitant to say that this is a software issue though because it isn't shown in any other flights, and the copy that I had downloaded from launch day doesn't have this error. The copy we put in the launch data folder does though. I'm pretty sure that someone just accidentally hit backspace and deleted the last digit, but it should be fixed because it's the one we have in the launch data folder.

[image]

[image]

wlsanderson, 2025-03-29:
delta time of IMU raw packet timestamp for pelicanator launch 1, in nanoseconds. 

[image]

harshil21, 2025-03-29:
the one around 5k lines up with the buffer. not sure what the spike at 12k is about. Also the one at 5k should just be a single line.

to be clear you're talking about the packet timestamp for both flights correct?

wlsanderson, 2025-03-29:
launch 2 is slightly more interesting. this is the dt of the imu packet timestamp after buffer (so x-axis is row# - 5000) This is plotting the dt between both raw and estimated. 

[image]

harshil21, 2025-03-29:
so we never really fixed the timestamp issues from legacy launch, where it also went negative. This seems to be very minor but still

wlsanderson, 2025-03-29:
I dont think this is the same issue as legacy. that was an internal imu buffer issue that was deleting data and we were somehow pulling old data I think. The negative values here are from one type of packets, either raw or est, getting ahead of the other packet type, causuing a negative dt when comparing raw time -> est time -> raw time -> est time, etc. When just comparing est timestamps, or just looking at raw timestamps, there are no negative dts. This might be an issue with the IMU internal queue, or an issue with how we are putting stuff in queues.

wlsanderson, 2025-03-29:
this also was not an issue with genesis launch 1 or 2

harshil21, 2025-03-29:
ok. I vaguely remember testing the IMU timestamp thing with you sometime after legacy (?) and it was fine then I think. At that point we had it at 1000Hz I think though. And I think that was just dt between any 2 packets, not specifically raw and est.

harshil21, 2025-04-12:
For anyone reading in the future: 

We debugged this and it turns out it's likely a power issue. The pelicanator flights used a Pi 5, which has a different policy on how much current to supply to devices. If you plug in the IMU to my laptop, the issue could not be replicated even at different frequencies. We noticed that changing raw frequencies to 500/1000 Hz with the IMU connected to the pi would sometimes correct the issue. 

It's also worth noting that setting the raw frequency to 500Hz still means that we are processing raw packets at 1000Hz because of the "gpsTimestamp" fields, which we can only properly tell the IMU not to send them from the code.

More thorough analysis would be needed, and I think that would involve using a USB multimeter to measure spikes in voltage/current when the dt spikes.